gmlarumbe/verilog-ext #15: Specify project directories with * glob syntax

abradd, 2024-01-11:
When defining `:dirs` in `verilog-ext-project-alist` you can use the `-r` flag to populate the list recursively, but there are situations where globbing may be preferred. For example I might want the following paths:

- `/proj_root/rtl/*/src`
- `/proj_root/rtl/*/tb`

Where the `*` replaces the name of a module.

I could achieve this with `:dirs` and `:ignore-dirs`, but depending on the size of the repository that could be tedious.

At the moment I use `file-expand-wildcards` with `*` globbing to populate project directories:

``` emacs-lisp
(setq my-verilog-dirs 
      (let ((value (list)))
        (dolist (element '("/proj_root/rtl/*/src" "/proj_root/rtl/*/tb") value)
          (setq value (append value (file-expand-wildcards element))))))
```

This could potentially be used alongside the existing `-r` flag.

An alternative to `-r` could be globstar (`**`): `/proj_root/rtl/**/src` so that multiple directory levels could be defined. I don't know how common very deep repositories are though. 

I couldn't find any built-in method to expand file paths with globstar. I've only seen references to using [fd](https://github.com/sharkdp/fd):

``` shell
fd -p -g '/root/rtl/**/src
```

This would introduce an external dependency.

Let me know what you think.

gmlarumbe, 2024-01-11:
Hi @abradd ,

Adding support for glob patterns in `verilog-ext-project-alist` would definitely make things easier and more flexible. 

I have implemented it in the `dev` branch (74722f9) for the `*` wildcard in case you want to test it and make sure it works in your environment before merging into main. It should work for all the properties (`:dirs`, `:ignore-dirs`, `:files` and `:ignore-files`) in `verilog-ext-project-alist` and also with the `-r` for `:dirs`. 

With respect to the globstar I could not find either any builtin in Emacs to expand file paths. I would prefer not to add more dependencies on external packages/binaries to make things simpler. However I assume that there will be use cases where it will come in handy. I have written a simple elisp function that should do the trick, even though it seems everything but efficient:
```elisp
(defun larumbe/globstar-dirs (globstar-dir &optional recursive)
  "Return list of directories that match GLOBSTAR-DIR expression.

If RECURSIVE is non-nil, return subdirectories recursively as well."
  (let* ((globstar-re "\\*\\*")
         (globstar-pos (string-match globstar-re globstar-dir))
         (subdirs-substring-start (1+ (string-match "/" globstar-dir globstar-pos)))
         (base-dir (substring-no-properties globstar-dir 0 globstar-pos))
         (predicate-dirs (string-trim-right (substring-no-properties globstar-dir subdirs-substring-start) "/+"))
         (filter-re (concat base-dir ".+" "\\<" predicate-dirs "\\>" (unless recursive "\\'")))
         all-dirs)
    (when globstar-pos
      ;; Keep only directories, discard files
      (setq all-dirs (seq-remove (lambda (x) (not (file-directory-p x)))
                                 (directory-files-recursively base-dir "." t)))
      ;; Return directories that match path after globstar
      (seq-filter (lambda (dir) (string-match-p filter-re dir)) all-dirs))))

;; Tested with the following code:
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/verilog-mode")
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/verilog-mode/")
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/verilog-mode///")
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/files")
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/files" :recursive)
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/snippets/") ; Should return nil, no intermediate dirs
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/git")
(larumbe/globstar-dirs "~/.emacs.d/straight/repos/verilog-ext/**/git" :recursive)
```
Let me know if it this function works for you or if on the contrary it takes ages to retrieve simple directories.

Thanks again for suggesting these improvements!

abradd, 2024-01-17:
> I have implemented it in the dev branch (https://github.com/gmlarumbe/verilog-ext/commit/74722f926b1b339f4f0d3ac0f97a01069b81e367) for the * wildcard in case you want to test it and make sure it works in your environment before merging into main. It should work for all the properties (:dirs, :ignore-dirs, :files and :ignore-files) in verilog-ext-project-alist and also with the -r for :dirs.

Initial testing of this on the `dev` branch looks good.

I haven't tested the globstar code yet.

gmlarumbe, 2024-01-17:
Sounds great! Let me know when you test the globstar before merging into main.

Thanks!

abradd, 2024-01-25:
I tested globstar using your examples and another repo I use. Works great.

gmlarumbe, 2024-01-26:
Nice! I will integrate it in the next few days when I spare some time.

There is one limitation I found that I will add to the documentation: globstar and `*` globbing cannot be used simultaneously for the same file/dir, e.g:
```elisp
"/proj_root/rtl/**/src/ip_*" ; Expansion returns nil
``` 

I hope this is enough for most use cases.

Thanks for testing!

closed 2024-02-12 by gmlarumbe