gmlarumbe/verilog-ext #18: Packages still download even though I comment them out...

whompyjaw, 2024-03-13:
I uncommented out `lsp-mode` and `flycheck` in the `verilog-ext-feature-list` but those package still download... That is counterintuitive... For very limited resources I have to be careful what packages I install.
```
(use-package verilog-ext
  :hook ((verilog-mode . verilog-ext-mode))
  :init
  ;; Can also be set through `M-x RET customize-group RET verilog-ext':
  ;; Comment out/remove the ones you do not need
  (setq verilog-ext-feature-list
        '(font-lock
          xref
          capf
          hierarchy
          eglot
          ;; lsp-mode
          beautify
          navigation
          template
          formatter
	  ;; flycheck
          compilation
          imenu
          which-func
          hideshow
          typedefs
          time-stamp
          block-end-comments
          ports))
  :config
  (verilog-ext-mode-setup)
  (verilog-ext-eglot-set-server 've-svls))
```

```
Building verilog-ext → Building lsp-mode...
Building verilog-ext → Building lsp-mode → Cloning f.el...done
Building verilog-ext → Building lsp-mode → Building f...
Building verilog-ext → Building lsp-mode → Building f → Cloning s.el...done
Building verilog-ext → Building lsp-mode → Building f → Building s...done
Building verilog-ext → Building lsp-mode → Building f...done
Building verilog-ext → Building lsp-mode → Cloning ht.el...done
Building verilog-ext → Building lsp-mode → Building ht...done
Building verilog-ext → Building lsp-mode → Cloning spinner...done
Building verilog-ext → Building lsp-mode → Building spinner...done
Building verilog-ext → Building lsp-mode → Cloning markdown-mode...done
Building verilog-ext → Building lsp-mode → Building markdown-mode...done
Building verilog-ext → Building lsp-mode → Cloning hydra (for lv)...done
Building verilog-ext → Building lsp-mode → Building lv...done
Building verilog-ext → Building lsp-mode...done
```
```
Building verilog-ext → Cloning flycheck...done
Building verilog-ext → Building flycheck...done
```

whompyjaw, 2024-03-13:
You also download `ag` and `ripgrep`... How come? Why not choose one or the other based on which binary is used? I exclusively use `ripgrep` and don't have `ag` in my path. Why would that package get installed?

gmlarumbe, 2024-03-16:
Hi @whompyjaw 

Thanks for opening an issue. I am trying to constantly improve the package and I appreciate having user feedback.

Even though you are commenting out the `flycheck` and `lsp-mode` features in the `:init` section of `use-package` elisp code, these are being downloaded because they belong to the `Package-Requires` header of the `verilog-ext` package. See https://www.gnu.org/software/emacs/manual/html_node/elisp/Library-Headers.html

As far as I know, package managers such as `package.el` or `straight.el` use this header to determine what are the dependencies of a package at download time. Thus downloading of these dependencies cannot be made conditionally. What is made conditionally with the `verilog-ext-feature-list` is the loading of these dependencies at run time. This prevents loading of unused features/packages and can help reducing Emacs startup time.

This is the best solution that I have thought about so far to make setup of the `verilog-ext` package as simple as possible for the average user. However as you have pointed out it relies on having downloaded all the dependencies/packages first.

The optimal solution would be to integrate the code of features such as `lsp-mode` or `flycheck` into their respective packages.  However this would require opening Pull Requests on their Git repos for each change, making development slower. Once `verilog-ext` proves to be mature enough I will try to make this integration.

Anyway if you can think of a better approach or solution I would be more than happy to try it.

Thanks!

whompyjaw, 2024-03-27:
Hi @gmlarumbe, I completely understand. Thanks for elaborating on the "behind the scenes". Your package is super helpful, so it's work eating that storage cost. It seems you have created the best solution, and if I *really* want to, I can remove those packages from that header file :)

And, sadly, I am not a proficient elisper (working on it), so I don't have any recommendations at this moment. I assume it's not possible to have the header as a modifiable variable? Maybe for the `:init` macro for use-package? If that's not feasible, feel free to close this, as your current solution seems to be the best!

gmlarumbe, 2024-03-28:
Hi @whompyjaw 

Thanks a lot for your understanding. I am happy to read that you find the package useful!

>  I assume it's not possible to have the header as a modifiable variable?

Some time ago I looked if that was possible but I did not find anything. The syntax of the `Package-Requires` field in the header seems very simple.

> Maybe for the :init macro for use-package?

I also looked at `straight.el` integration with `use-package` to see if that was possible, but could not find anything either...

I am closing the issue, but feel free to reopen if there is a better short/mid-term solution than the integration of `flycheck` or `lsp-mode` into their respective packages.

closed 2024-03-28 by gmlarumbe