gmlarumbe/verilog-ext #19: Can the project root be remote?

AlecVercruysse, 2024-03-28:
Hello, thank you for writing this package, I'm excited to try it out. I'm hoping to start by build a tags table for xref. When I run `verilog-ext-tags-get`, however, I get:
```
user-error: Not in a Verilog project buffer, check ‘verilog-ext-project-alist’
```
I've got a setup.el file for my project which looks like:
```
(setq verilog-ext-project-alist
      `(("project-name" ; Project name
         :root "/path/to/project/on/server"
         :files ("hardware/src/**/*.v"
                    "hardware/src/*.v")
         ))) 
```
I haven't looked behind the scenes but I'm guessing the fact that I'm using Tramp is messing things up. I've tried changing my root path to `ssh:<server>:/path/to/project/on/server` to no avail. Am I missing something or is this not currently possible? Thanks!

gmlarumbe, 2024-03-29:
Hi @AlecVercruysse ,

Thanks a lot for trying `verilog-ext` out! 

There are a couple of things to take into account:

- Currently `verilog-ext-tags-get` and the rest of commands that are project based require the user to be visiting a Verilog file from any of the projects in `verilog-ext-project-alist`.
- There are no regression tests for remote TRAMP cases. I never tried it before but I am glad you opened the issue since I found that it works! I think the only thing you might be missing is a slash at the beginning of your root path. This worked for me:
```elisp
(setq verilog-ext-project-alist
      `(;; Tramp test with glob
        ("verilog-ext tramp test" 
         :root "/ssh:<server>:/home/<user>/verilog-ext-tramp-test"
         :files ("*.sv"
                 "**/*.sv")
         )))
``` 
You can check if the package properly finds the project files first visiting the Verilog buffer and then running:
 - `M-: (verilog-ext-proj-files) RET`
 
It should return a list of strings with the absolute paths of the remote project. For example:
```elisp
("/ssh:<server>:/home/<user>/verilog-ext-tramp-test/file1.sv"
 "/ssh:<server>:/home/<user>/verilog-ext-tramp-test/file2.sv"
 "/ssh:<server>:/home/<user>/verilog-ext-tramp-test/subdir2/test3.sv"
 "/ssh:<server>:/home/<user>/verilog-ext-tramp-test/subdir1/test4.sv")
```

Finally, in case you are interested in knowing about the limitation of `verilog-ext-tags-get` requiring visiting a Verilog project file before executing, this is because the code uses this file to determine which project should build its tags table. I think this is useful if you are working on more than one project on a regular basis but on the other hand it could also be convenient to call it from outside of a Verilog buffer (e.g in a `dired` buffer). That could be an enhancement that would not require much effort. Maybe simply modifying the `verilog-ext-buffer-proj` function to prompt for a list of options with `completing-read` in case it returns nil. For example:
```elisp
(defun verilog-ext-buffer-proj ()
  "Return current buffer project if it belongs to `verilog-ext-project-alist'."
  (or (catch 'project
        (when (and buffer-file-name verilog-ext-project-alist)
          (dolist (proj verilog-ext-project-alist)
            (when (string-prefix-p (expand-file-name (plist-get (cdr proj) :root))
                                   (expand-file-name buffer-file-name))
              (throw 'project (car proj))))))
      (completing-read "Project: " (mapcar #'car verilog-ext-project-alist))))
```

If you or anyone else has some time to implement/test it and open a PR I would be more than happy. Otherwise I will take a look at this possibility in the future.

Thanks!

AlecVercruysse, 2024-03-30:
This is awesome, thank you! You're right that the fix was just the missing slash. I really appreciate that the tag generation with tree-sitter does not seem to require a remote executable, like I was experiencing with other options.

The current functionality of `verilog-ext-tags-get` is sufficient for my use-case, I only tend to regenerate tags while in a verilog buffer.

closed 2024-03-30 by AlecVercruysse