gmlarumbe/verilog-ext #22: Can't find definition

eychun, 2024-07-15:
I can't seem to use `M-x xref-find-definitions` to find the definition of reg "a" in the following code with Point at "a" on the 4th line:

> module tb;
>    reg[7:0] a;
>    initial begin
>       a = 8'hDA;
>       $display ("[%0t]: a = 0x%0h, b = 0x%0h, c = 0x%0h", $time, a, b, c);
>    end
> endmodule : tb

I get the following error message in Emacs:
> eglot--error: [eglot] Unsupported or ignored LSP capability `:definitionProvider'

I've done `M-x verilog-ext-tags-get RET` to get tags from the code.  I'm using Emacs 29.1.

Here are relevant lines from my .emacs file:
> (setq verilog-ext-feature-list
>       '(font-lock
>         xref
>         capf
>         hierarchy
>         eglot
>         lsp
>         flycheck
>         beautify
>         navigation
>         template
>         formatter
>         compilation
>         imenu
>         which-func
>         hideshow
>         typedefs
>         time-stamp
>         block-end-comments
>         ports))
> (require 'verilog-ext)
> (verilog-ext-mode-setup)
> (add-hook 'verilog-ts-mode-hook #'verilog-ext-mode)
> (verilog-ext-eglot-set-server 've-svls) ;`eglot' config
> (setq verilog-ext-project-alist
>       `(("training" ; Project name
>          :root "/home/eychun/training" ; supports remote dirs via Tramp
>          :files ("*.v") ; Multiple files can be specified through the glob pattern
>          :compile-cmd "make tb_top" ; command used to compile current project
>          :lib-search-path nil)))    ; list of dirs to look for include directories or libraries
> (setq verilog-ext-tags-backend 'tree-sitter)

Here is what I have in my .svls.toml file:
> [verilog]
> include_paths = ["blah.sv"]
> defines = ["DEBUG", "VAR=1"]
> 
> [option]
> linter = true

Am I missing any steps?

gmlarumbe, 2024-07-26:
Hi @eychun ,

Thanks for opening an issue.

The function `xref-find-definitions` uses only one of the available backends, which in your case seem to be the `verilog-ext` backend (since you have xref included in `verilog-ext-feature-list`) and `eglot` backend (through LSP server `svls`).

It seems to me that for some reason the `verilog-ext` xref backend is not active. Can you check the value of the variable `xref-backend-functions`? Its first element should be `verilog-ext-xref-backend`. Besides, can you also check if tags have been collected properly on the files in directory `~/.emacs.d/verilog-ext` ?

On the other hand, the `svls` language server only supports diagnostics but not finding definitions, so that is the reason why you are getting the error:

> eglot--error: [eglot] Unsupported or ignored LSP capability `:definitionProvider'

eychun, 2024-07-27:
Hi @gmlarumbe,

Thank you for your reply to my post.  Here is the value of the variable `xref-backend-functions`:
> xref-backend-functions is a variable defined in ‘xref.el’.
> 
> Its value is
> (eglot-xref-backend verilog-ext-xref-backend etags--xref-backend)
> Local in buffer blah.sv; global value is 
> (etags--xref-backend)

I checked if tags have been collected properly.  Here is what I see in `~/.emacs.d/verilog-ext/defs-file-tables`:
[image]

By the way, I have a question about something you said:
> the svls language server only supports diagnostics but not finding definitions

How did you find this information?  I couldn't find it in the https://github.com/dalance/svls/blob/master/README.md.

eychun, 2024-08-03:
Thank you.  I'm able to jump to definitions now.

[image]

After setting up eglot to not use xref, here is the value of `xref-backend-functions`:

> Its value is (verilog-ext-xref-backend etags--xref-backend)
> Local in buffer blah.sv; global value is 
> (etags--xref-backend)

I'm a little confused by all the pieces necessary for using LSP.  I think svls is a language server.  Is verilog-ext also a language server?  svls works with eglot.  Does verilog-ext also work with Eglot or does it work by itself?

gmlarumbe, 2024-08-04:
Hi @eychun ,

`verilog-ext` is an Emacs package that extends the functionality of `verilog-mode` and `verilog-ts-mode` to edit SystemVerilog code and provides similar features as LSP does (e.g. find definition).

On the other hand for LSP you need a server (e.g. `svls`) and a client (e.g. Emacs `eglot` package). You could configure `eglot` to connect to `svls` without the need of `verilog-ext`. `verilog-ext` simply provides some basic code to simplify the process for all the current known LSP servers and clients for Verilog.

You can check the wiki entry for more info: https://github.com/gmlarumbe/verilog-ext/wiki/Language-Server-Protocol

closed 2024-08-04 by gmlarumbe