gmlarumbe/verilog-ext #23: Incorrect indentation of directives in verilog-mode

punzik, 2024-07-24:
When the verilog-ext-mode is activated, indents before directives are broken. The `verilog-calculate-indent` function from the verilog-mode starts returning an incorrect indentation type.

init.el
```elisp
(require 'package)
(add-to-list 'package-archives '("melpa" . "https://melpa.org/packages/"))

(use-package verilog-mode
  :ensure t)

(use-package verilog-ext
  :ensure t
  :config 
  (verilog-ext-mode-setup))
```

Indentation without verilog-ext-mode:
```verilog
`define MACRO_1
`define MACRO_2

module test;
endmodule
```

Indentation with verilog-ext-mode:
```verilog
`define MACRO_1
  `define MACRO_2

    module test;
endmodule
```

GNU Emacs 29.4 (build 1, x86_64-pc-linux-gnu, GTK+ Version 3.24.42, cairo version 1.18.0)

gmlarumbe, 2024-07-26:
Hi @punzik ,

Thanks for opening an issue. I have been able to reproduce it under `verilog-mode` once `verilog-ext-mode` is first enabled and then disabled. Is that the case for you as well?

If so, the reason seems to be the override of the syntax-table in `verilog-ext-mode` minor mode definition, in `verilog-ext.el` file:
```elisp
        (when (eq major-mode 'verilog-mode)
          ;; Syntax table overriding:
          ;; Avoid considering tick as part of a symbol on preprocessor directives while
          ;; isearching.  Works in conjunction with `verilog-ext-tab'
          ;; and `verilog-ext-indent-region' to get back standard table to avoid
          ;; indentation issues with compiler directives.
          (modify-syntax-entry ?` ".")
```
I have tried locally with the following patch and works for me after disabling `verilog-ext-mode`. Could you also test it locally?
```diff
diff --git a/verilog-ext.el b/verilog-ext.el
index abeab05..73383f4 100644
--- a/verilog-ext.el
+++ b/verilog-ext.el
@@ -300,6 +300,8 @@ FEATURES can be a single feature or a list of features."
           (verilog-ext-when-feature 'which-func
             (verilog-ext-which-func))))
     ;; Cleanup
+    (when (eq major-mode 'verilog-mode)
+      (modify-syntax-entry ?` "w")) ; Set back original syntax-table
     (remove-hook 'kill-buffer-hook #'verilog-ext-kill-buffer-hook :local)
     (verilog-ext-when-feature 'xref
       (verilog-ext-xref-set :disable))
```

If you also want to contribute with a PR that would be highly appreciated since I might not be available during summer and CI tests would run automatically on those changes.

Thanks a lot!

punzik, 2024-07-27:
Thank you so much for your help, @gmlarumbe! 

Now I understood what the problem was. The fact is that I do not use `verilog-ext-mode` key bindings (when I turn on the `verilog-ext-mode`, I do `(defvar verilog-text-mode-map nil)`). Therefore, I use `verilog-mode` functions for `TAB` and `Enter`, which behave incorrectly.

To fix this, I set the bindings from `verilog-ext`, and replaced the `indent-region` function with `verilog-ext-indent-region`. This change made everything work correctly.

If you don't mind, I'll close the issue.

Your correction fixed the erroneous behavior when `verilog-ext-mode` disabled. I think it should be added to the code.

closed 2024-07-27 by punzik

gmlarumbe, 2024-07-29:
Glad it worked! :)

> Your correction fixed the erroneous behavior when verilog-ext-mode disabled. I think it should be added to the code.

Added in commit de2f4af