gmlarumbe/verilog-ext #25: verilog-ext-beautify-xxx is ugly.

ziven123, 2025-02-19:
I can't show this beautiful code style by verilog-ext-beautify-xxx Command. Can you tell how to it work?

[image]

ziven123, 2025-02-19:
When I use C-M-i, My code format is same as verilog-mode and this code format very ugly.
I want the code style is like your picture.

gmlarumbe, 2025-02-19:
Hi @ziven123 ,

Thanks for opening an issue. I have a couple of questions:
- Are you using `verilog-mode` or `verilog-ts-mode` ?
- What is exactly the code snippet you are trying to beautify, the result you get and the one that you expect? Do you refer to the indentation of ports in instances?
- What is the value of `verilog-indent-lists`, in case you are using `verilog-mode`?

Thanks!

ziven123, 2025-02-20:
@gmlarumbe 
I am using verilog-ext nad verilog-mode.
Origin code format is below,
[image]

When I input C-c C-b verilog-ext-beautify-current-buffer, it is ugly format below,
[image]

I hope the code format can be as beautiful as yours.
[image]

Additional, I can't found verilog-indent-lists.
[image]

thanks
Ziven

ziven123, 2025-02-21:
Hi @gmlarumbe，

Thanks for your reply. I set verilog-indent-list to nil, it show below,
[image]

But all ")" is not align.
There is no space between any '(' or ')' and the string.
I still hope the code format can be as beautiful as yours
[image]

Thanks
Zhiwen

ziven123, 2025-02-24:
Hi @gmlarumbe ,

Thank for your reply.
I am going to find other tool.

Thanks again,
Ziven

gmlarumbe, 2025-02-25:
Hi @ziven123 ,

The instantiation format that you suggested has been implemented in [f742d53](https://github.com/gmlarumbe/verilog-ext/commit/f742d531412733b0217147eb862fecfa09fa3753).

If using `verilog-mode`:
```elisp
(setq verilog-ext-beautify-instance-extra t)
```

If using `verilog-ts-mode`:
```elisp
(setq verilog-ts-beautify-instance-extra t)
```

closed 2025-02-28 by gmlarumbe