gmlarumbe/verilog-ext #29: Issue with Code Beautify

Remillard, 2025-07-16:
I have run into an issue with the code beautification routine.  It seems like it does not ignore comments in instantiations when lining up named instantiation assignments.  In particular if you have a comment that includes a parenthetical, it'll try to align the comment AND it will take the furthest right open paren as the starting point for the column.

For example:
```
foo_module #(
    .G_SOME_CONSTANT (5)
) my_foo (
    .clk (my_clk),
    .reset (my_reset)

    // Comment regarding the ports below (because sometimes it's warranted)
    .port_a (my_a),
    .port b (my_b)

    //Another comment (with paren)
    .port_c (my_c),
    .port_d (my_d)
);
```
will beautify to:
```
foo_module #(
    .G_SOME_CONSTANT (5)
) my_foo (
    .clk                                 (my_clk),
    .reset                               (my_reset)

    // Comment regarding the ports below (because sometimes it's warranted)
    .port_a                              (my_a),
    .port b                              (my_b)

    //Another comment                    (with paren)
    .port_c                              (my_c),
    .port_d                              (my_d)
);
```

Note that it chose the paren in the comment to align with, and additionally realigned the comment as well!  Thus I think it's reading comment lines and considering them to be valid targets for measuring port parenthesis distance.

gmlarumbe, 2025-07-17:
Hi @Remillard ,

Thanks for reporting the issue and doing some research on it.

As you mentioned it is caused by the parenthesis in the comment which should not be taken into account. However with current implementation using `align-regexp` this is not possible.

I will try to implement the beautifying with a different approach that ignores comments to avoid this type of issue.

Remillard, 2025-07-17:
For that matter, if you know Javascript, there's a JS Verilog code beautifier out there too that might have a good strategy.  I haven't used it though.  Verible's code beautifier is only okay -- it does some things with task/function arguments that I really hate but otherwise is acceptable.